GyverLibs/FastBot2 #1: Ошибка парсинга при командах

Er2n, 2023-12-31:
Лонг пул потрясающий. Самое то для проектов где команды редко прилетают и не хочется сильно сеть грузить. Но либа пока очень сырая, например при отправке в бота команды со слешом, ошибка парсинга прилетает.

GyverLibs, 2023-12-31:
Все проблемы списком пожалуйста, будут быстрее решаться))

Er2n, 2023-12-31:
Пока только это нашел, ну и еще skipUpdates не все обновления пропускает, оставляет одно последнее. Но это хз как фиксить, если тг так реагирует на offset=-1.

GyverLibs, 2023-12-31:
Да, оффсет -1 не всё скидывает, подарки от Пашки

GyverLibs, 2023-12-31:
с командой проблема в GSON библиотеке, сейчас зафикшу

Er2n, 2023-12-31:
[image]
Ему нолик не нравится.

GyverLibs, 2023-12-31:
я знаю, поэтому и написал сейчас пофикшу)

Er2n, 2023-12-31:
Крутяк, спасибо за скоростное исправление))

closed 2023-12-31 by Er2n