GyverLibs/FastBot2 #5: Как использовать setMyCommands ?

AndyB117, 2024-04-10:
В FastBot и UniversalBot использовал 
```c++
bot.setMyCommands(
    "["
    "{\"command\":\"help\", \"description\":\"Помощь по командам\"},"
    "{\"command\":\"info\", \"description\":\"Информация о настройках\"},"
    "{\"command\":\"status\",\"description\":\"Показания датчиков\"}," 
    "{\"command\":\"restart\",\"description\":\"Перезагрзка контроллера\"}"     // no comma on last command
    "]"
  );
```
В FastBot2 пытался
```c++
fb::Packet p = bot.beginPacket(fb::cmd::setMyCommands);
	p.addString(fb::api::commands, 
//		"commands="
    	"["
    	"{\"command\":\"help\", \"description\":\"Помощь по командам\"},"
    	"{\"command\":\"info\", \"description\":\"Информация о настройках\"},"
    	"{\"command\":\"status\",\"description\":\"Показания датчиков\"}," 
    	"{\"command\":\"restart\",\"description\":\"Перезагрзка контроллера\"}"     // no comma on last command
    	"]"
	);
```
и
```c++
p = bot.beginPacket(fb::cmd::setMyCommands);
p.addStringRaw (
		"?commands="
    	"["
...
```
и
```c++
p = bot.beginPacket(fb::cmd::setMyCommands);
p.addText (
		"commands="
    	"["
...
```
Все равно выдается ошибка от сервера, мол не могу распарсит JSON.
А как правильно?

GyverLibs, 2024-04-10:
Здесь параметры передаются в json, а не в query string. Знак = не нужен, посмотри как в примере идёт работа с packet или в любом месте в библиотеке

closed 2024-04-10 by GyverLibs

AndyB117, 2024-04-10:
Так в примере просто посылается Message. И все.
```c++
fb::Packet p = bot.beginPacket(fb::cmd::sendMessage);  // Все команды API Telegram доступны в fb::cmd
 p.addString(fb::api::text, "message text");  // все ключи объектов API Telegram доступны в fb::api
    // p.addInt(fb::api::chat_id, CHAT_ID);
bot.sendPacket(p);
```

GyverLibs, 2024-04-10:
я скину пару примеров, сейчас проверяю на железе. Ошибки сервера нет, но и команды не появляются

GyverLibs, 2024-04-10:
в общем так
```cpp
p.beginArr(fb::api::commands);
p.addText(R"(
    {"command":"help","description":"Помощь по командам"},
    {"command":"info","description":"Информация о настройках"},
    {"command":"status","description":"Показания датчиков"},
    {"command":"restart","description":"Перезагрзка контроллера"}
)");
p.endArr();
```
или нативно по одной
```cpp
p.beginArr(fb::api::commands);
p.beginObj().addString(fb::api::command, "help").addString(fb::api::description, "Помощь по командам").endObj();
p.beginObj().addString(fb::api::command, "info").addString(fb::api::description, "Информация о настройках").endObj();
p.endArr();
```

AndyB117, 2024-04-10:
Спасибо, сейчас попробую.

AndyB117, 2024-04-10:
Хм, не появляется. Перезапуск телеги - это перезапуск приложения, очистка и /start в боте по-новом или revoke key?

GyverLibs, 2024-04-10:
перезапуск приложения да

AndyB117, 2024-04-10:
вот фрагменте кода
```c++
fb::Packet p;
	p = bot.beginPacket(fb::cmd::setMyCommands);
	p.beginArr(fb::api::commands);
	p.addText(R"(
    	{"command":"help","description":"Помощь по командам"},
    	{"command":"info","description":"Информация о настройках"},
    	{"command":"status","description":"Показания датчиков"},
		{"command:""tpar","description":"Заданная температура парилки"}, 
    	{"command":"restart","description":"Перезагрзка контроллера"}
	)");
	p.endArr();
//	p.addInt(fb::api::chat_id, CHAT_ID);
	FB_LOG("" + p.s + "\n");
    bot.sendPacket(p, false);
```
А вот полученный запросзапрос
```
14:23:31.889 > > GET /bot6995012195:AAE7PHsaJ7y9gLpIhejA07hn0sOj8q61MXQ/setMyCommands HTTP/1.1
14:23:31.889 > Host: api.telegram.org
14:23:31.889 > Content-Type: application/json
14:23:31.889 > Content-Length:      
14:23:31.889 > 
14:23:31.889 > {,"commands":[
14:23:31.889 >      {"command":"help","description":"Помощь по командам"},
14:23:31.889 >      {"command":"info","description":"Информация о настройках"},
14:23:31.889 >      {"command":"status","description":"Показания датчиков"},
14:23:31.889 >      {"command:""tpar","description":"Заданная температура парилки"}, 
14:23:31.889 >      {"command":"restart","description":"Перезагрзка контроллера"}
14:23:31.889 >  ],
14:23:31.889 >  in bot_setup() [src/telegram.cpp : 81]
14:23:31.889 > [  1106][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
14:23:32.147 > {"ok":true,"result":true}
```
{,"commands":[

Здесь запятая не лишняя?

GyverLibs, 2024-04-10:
возможно стоит обновить gson/stringutils до актуальных версий

AndyB117, 2024-04-10:
Обновил, с Гита. Все тоже.
На пару часов отложу тесты, поработать просят )))

GyverLibs, 2024-04-10:
так, проблема в 
```
fb::Packet p = bot.beginPacket(fb::cmd::sendMessage);
```
и
```
fb::Packet p;
p = bot.beginPacket(fb::cmd::setMyCommands);
```

GyverLibs, 2024-04-10:
второй вариант даёт лишнюю запятую, буду фиксить

GyverLibs, 2024-04-10:
нужен gson 1.4.10

GyverLibs, 2024-04-10:
Мой пример заработал?

AndyB117, 2024-04-11:
Перебил названия, на которые ругался компилятор. Скомпилировалось.
Но меню появляется только после перезапуска телеги.

> 
> 
>Мой пример заработал?
 
 
С уважением,
Andy Neverowsky
***@***.***

AndyB117, 2024-04-11:
Там 2  ошибки опечатки у тебя  в коде. Вместо латинской стоят русские «С»
Из патчика понятно.

> 
> 
>Мой пример заработал?
 
 
С уважением,
Andy Neverowsky
***@***.***

AndyB117, 2024-04-11:
Похоже дело сейчас уже не в боте.
Попробовал сделать такой запрос:
TELEGRAM_TOKEN=XXXXXX:YYYYYYYYYYYYYYY
curl -X POST -H "Content-Type: application/json" -d "{\"commands\":[{\"command\":\"test\",\"description\":\"description with spaces\"}]}"  https://api.telegram.org/bot$TELEGRAM_TOKEN/setMyCommands
 
тоже меню появляется, но не сразу, а после загрузки.
А вот в методе GET, который используется в FastBot (первом), все появляется сразу.
Скорее всего нужно еще что то послать, типа какого-нибудь refresh (telegram API не знаю).
Надо будет посмотреть, что там в UniversalTelegramBot. Но это уже после работы и то если успею.

> 
> 
>Мой пример заработал?
 
 
С уважением,
Andy Neverowsky
***@***.***

AndyB117, 2024-04-11:
Чтоб добить тему.
В общем заработал вариант, в котором элементы меню добавляются по одному.
fb :: Packet p = bot . beginPacket ( fb :: cmd :: setMyCommands );
p . beginArr ( fb :: api :: commands );
p . beginObj (). addString ( fb :: api :: command ,  "help" ). addString ( fb :: api :: description ,  "Помощь по командам" ). endObj ();
p . beginObj (). addString ( fb :: api :: command ,  "info" ). addString ( fb :: api :: description ,  "Информация о настройках" ). endObj ();
p . beginObj (). addString ( fb :: api :: command ,  "status" ). addString ( fb :: api :: description ,  "Показания датчиков" ). endObj ();
p . beginObj (). addString ( fb :: api :: command ,  "restart" ). addString ( fb :: api :: description ,  "Перезагрзка контроллера" ). endObj ();
p . endArr ();
bot . sendPacket ( p );
 
Когда сразу все меню добавляется — не заработал.
Но есть нюанс, который немного сбил с понталыку.
Поскольку изначально меню не появлялось, я в сетапе отправлял приветствие в чат. После запуска бота приходило уведомление на телефон и когда поднимал телефон, то сразу оказывался в чате. Меню не было. Но, если выйти из чата и зайти, то меню появлялось.
Убрал приветствие, теперь запускаю бота, захожу в чат — меню на месте.
 
Спс. Это уже рабочий вариант.

> 
> 
>Мой пример заработал?
 
 
С уважением,
Andy Neverowsky
***@***.***

GyverLibs, 2024-04-11:
лол, а у меня не ругается на кириллицу (stray \320). И вот этот работает ок
```
p.beginArr(fb::api::commands);
p.addText(R"(
    {"command":"help","description":"Помощь по командам"},
    {"command":"info","description":"Информация о настройках"},
    {"command":"status","description":"Показания датчиков"},
    {"command":"restart","description":"Перезагрзка контроллера"}
)");
p.endArr();
```

GyverLibs, 2024-04-11:
да, есть там ещё одна русская) сейчас допишу доку ещё, запушу