GyverLibs/FastBot2 #6: Ошибка CORRUPT HEAP на esp wroom 32

DrGames580501, 2024-05-02:
Использовал скетч test.ino из папки с примерами. Через пару секунд после подключения к wifi возникает следующая ошибка:

CORRUPT HEAP: Bad tail at 0x3ffc92c5. Expected 0xbaad5678 got 0x226b6f22

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40083669:0x3ffb2000 0x4008bbf9:0x3ffb2020 0x40091079:0x3ffb2040 0x40090c71:0x3ffb2170 0x40083bd9:0x3ffb2190 0x400910a9:0x3ffb21b0 0x40153c4d:0x3ffb21d0 0x40153b01:0x3ffb21f0 0x400d5be9:0x3ffb2210 0x400d5cc3:0x3ffb2250 0x400d5d0f:0x3ffb2270 0x400db299:0x3ffb2290

ELF file SHA256: fe90fcaeab059855

Rebooting...

После чего плата перезагружается, и происходит всё тоже самое, что я написал ранее. Единственное что я заметил, при добавлении delay в loop эта ошибка возникает реже. На этом всё, буду рад любой помощи

GyverLibs, 2024-05-02:
токен бота установлен? Также можно включить 
```
#define FB_USE_LOG Serial
```
перед подключением библиотеки, в сериал будут выводиться ошибки

DrGames580501, 2024-05-03:
Спасибо. Токен установлен, вывод в сериал попробую чуть позже установить

DrGames580501, 2024-05-03:
Я ничего не менял и попробовал снова. Всё работало нормально, только выскочила ошибка wifi:Set status to INIT, но это, как я понял, уже мои проблемы. Продолжу экспериментировать, и если всё будет работать нормально, закрою тему и буду искать проблему в плате

GyverLibs, 2024-05-03:
Окей

DrGames580501, 2024-05-04:
Короч видимо проблема в плате. Я проверил большую часть необходимых мне функций и всё работало нормально. В сериал ничего подозрительно не выводилось

closed 2024-05-04 by DrGames580501

GyverLibs, 2024-05-04:
шик 😎

GyverLibs, 2024-05-04:
буду рад предложениям/пожеланиям и репортам о сбоях. Вроде написал всё что хотел, но не тестировал особо. Собираю фидбеки, надо убедиться что всё хорошо работает и можно релизить)