GyverLibs/FastBot2 #7: editMessageCaption (editCaption)

impulseDiMaro, 2024-06-04:
Доброе время суток.
не удается отредактировать заголовок Inline Menu
в первой версии это прекрасно работает вот так

> if (m_method) bot.inlineMenuCallback("Старый заголовок", str_menu, str_call, id_chat);
  else bot.editMessageMenuCallback(menuID, "Новый заголовок", str_menu, str_call, id_chat);

во второй версии создаю inline menu вот так (работает)

> fb::Message msg(head[head_menu], id_chat);
fb::InlineMenu menu(str_menu, str_call);
msg.setInlineMenu(menu);
bot.sendMessage(msg, true);
menuID = bot.lastBotMessage();

редактирую inline menu (и это работает)

> fb::MenuEdit edit(menuID, id_chat);
fb::InlineMenu menu(str_menu, str_call);
edit.setInlineMenu(menu);
bot.editMenu(edit);

пытаюсь отредактировать заголовок, ибо без этой возможности инлайн меню теряет всякий смысл, во всяком случае при построении многоуровневого динамического меню
так

> fb::CaptionEdit capt;
capt.chatID = id_chat;
capt.caption = "New Caption";
capt.messageID = menuID;
bot.editCaption(capt);

или так

> fb::CaptionEdit capt("New Caption", menuID, id_chat);
bot.editCaption(capt);

из-за проблемы пытаюсь смотреть что улетает серверу

> POST https://api.telegram.org/bot62xx453xxx:AAT-tXR-HQN_R-ggfvjArzVr7aA-h2u-vjo/editMessageCaption HTTP/1.1
Host: api.telegram.org
Cache-Control: no-cache
Content-Type: application/json
Content-Length:      
{"chat_id":136xx15xxx,"message_id":2984,"caption":"New Caption",

тут у меня обрезается JSON в логе, и это не ограничение размера, куда более емкие проходят, но все обрезанные
ID сообщения меню корректное
сервер всегда возвращает ошибку

> {"ok":false,"error_code":400,"description":"Bad Request: there is no caption in the message to edit"}

GyverLibs, 2024-06-04:
Привет, caption относится к сообщениям с файлами, сообщение с меню это TextEdit, отправлять через editText

impulseDiMaro, 2024-06-04:
удивительно, через 5 минут после отправки своего сообщения сам допер до этого ))))
пришел извиниться, мол "сам дурак",  а тут ответ уже.. не ожидал даже. спасибо за ответ Алекс!

GyverLibs, 2024-06-04:
Пожс)

closed 2024-06-04 by GyverLibs