GyverLibs/FastBot2 #9: Ошибка компиляции

woodlist, 2024-08-13:
Espressif обновился на версию 5.1 и обновил ардуино ядро до 3.0.4 версии.
Борд менеджер ссылается на:
`https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json
https://espressif.github.io/arduino-esp32/package_esp32_index.json`
Там перебрали все кардинально и теперь почти все сторонние библиотеки встали.
`C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:49:6: error: variable or field 'rawh' declared void
   49 | void rawh(Text text) {
      |      ^~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:49:11: error: 'Text' was not declared in this scope
   49 | void rawh(Text text) {
      |           ^~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:53:6: error: variable or field 'handleCommand' declared void
   53 | void handleCommand(fb::Update& u) {
      |      ^~~~~~~~~~~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:53:20: error: 'fb' has not been declared
   53 | void handleCommand(fb::Update& u) {
      |                    ^~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:53:32: error: 'u' was not declared in this scope
   53 | void handleCommand(fb::Update& u) {
      |                                ^
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:218:6: error: variable or field 'handleMessage' declared void
  218 | void handleMessage(fb::Update& u) {
      |      ^~~~~~~~~~~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:218:20: error: 'fb' has not been declared
  218 | void handleMessage(fb::Update& u) {
      |                    ^~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:218:32: error: 'u' was not declared in this scope
  218 | void handleMessage(fb::Update& u) {
      |                                ^
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:256:6: error: variable or field 'handleDocument' declared void
  256 | void handleDocument(fb::Update& u) {
      |      ^~~~~~~~~~~~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:256:21: error: 'fb' has not been declared
  256 | void handleDocument(fb::Update& u) {
      |                     ^~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:256:33: error: 'u' was not declared in this scope
  256 | void handleDocument(fb::Update& u) {
      |                                 ^
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:284:6: error: variable or field 'handleQuery' declared void
  284 | void handleQuery(fb::Update& u) {
      |      ^~~~~~~~~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:284:18: error: 'fb' has not been declared
  284 | void handleQuery(fb::Update& u) {
      |                  ^~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:284:30: error: 'u' was not declared in this scope
  284 | void handleQuery(fb::Update& u) {
      |                              ^
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:312:6: error: variable or field 'updateh' declared void
  312 | void updateh(fb::Update& u) {
      |      ^~~~~~~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:312:14: error: 'fb' has not been declared
  312 | void updateh(fb::Update& u) {
      |              ^~
C:\Users\woodl\AppData\Local\Temp\.arduinoIDE-unsaved2024713-19340-5cmbg4.kbcw9\test\test.ino:312:26: error: 'u' was not declared in this scope
  312 | void updateh(fb::Update& u) {
      |                          ^

exit status 1

Compilation error: variable or field 'rawh' declared void`

GyverLibs, 2024-08-13:
Отличный повод пока не переходить на 3.х :)

woodlist, 2024-08-13:
В железе пока не проверял, но компиляция пошла без ругани и мата на библиотеке [https://github.com/cotestatnt/AsyncTelegram2](url)

GyverLibs, 2024-08-13:
я хз, вот на самой свежей версии esp32 всё компилится отлично, проблема где то в другом месте явно

[image]

[image]

closed 2024-08-13 by GyverLibs

woodlist, 2024-08-13:
[Compile_verbose.txt](https://github.com/user-attachments/files/16604266/Compile_verbose.txt)
Вот пожалуйста, сам черный ящик.

GyverLibs, 2024-08-13:
чёрный ящик пишет что не объявлены функции, это либо бесполезная информация, либо правда скетч кривой

woodlist, 2024-08-13:
Это был собственный скетч из примеров библиотеки. Я ничего своего не внёс там.

GyverLibs, 2024-08-13:
поставил arduino ide, в ней скетч не собирается из за объявленных наверху функций, если их закомментировать то всё ок. Сегодня добавлю простых примеров, а то их оказывается нету