GyverLibs/FastBot2 #10: Коллбэки и std::bind

LittleBuster, 2024-08-27:
Доброго времени суток. Огромное спасибо за возможность управлять контроллерами с телеги.

Подскажите как забиндить коллбэки бота к членам класса?
Пробобую так - не выходит

```
class TgBot
{
private:
    FastBot2    *_fb;
public:
    TgBot(FastBot2 *fb)
    {
        _fb = fb;
    }

    void updateHandler(fb::Update& u)
    {
    }

    void begin()
    {
        _fb->attachUpdate(std::bind(&TgBot::updateHandler, this, std::placeholders::_1));
    }
};
```

в core.h заменил
```
typedef void (*CallbackUpdate)(Update& upd);
```
на
```
typedef std::function<void(Update&)> CallbackUpdate;
```
лог
```
In file included from src/main.cpp:23:
include/net/tgbot/tgbot.hpp: In member function 'void TgBot::begin()':
include/net/tgbot/tgbot.hpp:81:88: error: no matching function for call to 'FastBot2::attachUpdate(std::_Bind_helper<false, void (TgBot::*)(fb::Update&), TgBot*, const std::_Placeholder<1>&>::type)'
         _fb->attachUpdate(std::bind(&TgBot::updateHandler, this, std::placeholders::_1));
                                                                                        ^
In file included from .pio/libdeps/esp32_r8n16/FastBot2/src/FastBot2Client.h:2,
                 from .pio/libdeps/esp32_r8n16/FastBot2/src/FastBot2.h:13,
                 from include/net/tgbot/tgbot.hpp:4,
                 from src/main.cpp:23:
.pio/libdeps/esp32_r8n16/FastBot2/src/core/core.h:125:10: note: candidate: 'void fb::Core::attachUpdate(fb::Core::CallbackUpdate)'
     void attachUpdate(CallbackUpdate callback) {
          ^~~~~~~~~~~~
.pio/libdeps/esp32_r8n16/FastBot2/src/core/core.h:125:10: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (TgBot::*)(fb::Update&), TgBot*, const std::_Placeholder<1>&>::type' {aka 'std::_Bind<void (TgBot::*(TgBot*, std::_Placeholder<1>))(fb::Update&)>'} to 'fb::Core::CallbackUpdate' {aka 'void (*)(fb::Update&)'}
```

GyverLibs, 2024-08-27:
видимо разбираться как работает std bind

GyverLibs, 2024-08-27:
я добавил фичу, глобально определён `fb::thisBot` - указатель на текущий объект, который вызвал коллбэк. Этого должно хватить. Типа
```cpp
void update(fb::Update& u) {
    static_cast<FastBot2*>(fb::thisBot)->sendMessage(...);
}
```

closed 2024-08-27 by GyverLibs