GyverLibs/FastBot2 #12: OTA error

OleksandrGora, 2024-08-29:
Sometimes I get OTA error message on ESP32 devkit.
The complicated code is the more often the message appears. The file size is not changed.
Partition map is not changed. It has been uploaded over a cable at the very beginning.

Where to dig?

GyverLibs, 2024-08-29:
bot ticks on default loop task or custom rtos task? Your update handler is a big single function or splitted to little ones as shown in example/docs?

OleksandrGora, 2024-08-29:
Loops and update funks look like that:
``` 
void loop()
{
  loop_firebase();
  loop_fastbot();
  loop_timers();
  loop_touch_check();
}

void loop_fastbot()
{
    bot.tick();
}

void update_bot(fb::Update &u)
{
    if (u.isMessage() &&
        u.message().hasDocument() &&
        u.message().document().name().endsWith(".bin") &&
        u.message().from().id() == CHAT_ID &&
        u.message().document().name().indexOf(BOARD) > -1)
    {
        fb::Fetcher fetch = mounting_bot.downloadFile(u.message().document().id());
        // OTA
        bool ok = fetch.updateFlash();
        bot.sendMessage(fb::Message(ok ? "OTA done" : "OTA error", u.message().chat().id()));
    }
}
```
List of libs used in the project:
gyverlibs/FastBot2@^1.0.2
	gyverlibs/UnixTime@^1.1
	gyverlibs/GyverNTP@^1.3.1
	beegee-tokyo/DHT sensor library for ESPx@^1.19
	mobizt/FirebaseClient@^1.3.10
	sstaub/TickTwo@^4.4.0

GyverLibs, 2024-08-29:
seems like the board is rebooting, you should check serial monitor for exceptions

GyverLibs, 2024-08-29:
i will make new way to OTA today, it will update from main loop to decrease memory using etc

OleksandrGora, 2024-08-29:
The board is not rebooting.
It does not freeze as well. It works as usual. At the moment there is no messages in serial that could help to trace the error.
Sometimes rebooting happens, but it is not in this case.

Partition - if it can help:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     ,        0x2000,
app0,     app,  ota_0,   ,        0x1E0000,
app1,     app,  ota_1,   ,        0x1E0000,
spiffs,   data, spiffs,  ,        0x20000,
coredump, data, coredump,,        0x10000,

GyverLibs, 2024-08-29:
uncomment [this](https://github.com/GyverLibs/FastBot2/blob/7f4ef23caeb08f45f37ef854b503d4be54fc8b53/src/bot_config.h#L21) and try ota

GyverLibs, 2024-08-29:
in new update you can use
```
bot.updateFlash(u.message().document(), u.message().chat().id());
```

OleksandrGora, 2024-08-29:
After uncommenting got this:
> getFile in downloadFile() [.pio/libdeps/esp32/FastBot2/src/FastBot2Client.h : 76]
> send + wait in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 267]
> bad response in _parseResponse() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 325]
> send + wait in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 267]
> got json in _parseResponse() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 307]
> got result in _parseResult() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 331]
> getUpdates in getUpdates() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 208]
> send async in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 270]

OleksandrGora, 2024-08-29:
After update to 1.0.3 OTA done! Thanks!

Closing the issue.

GyverLibs, 2024-08-29:
1.0.3? Nice

OleksandrGora, 2024-08-30:
Unfortunately OTA does not seem to work. Yesterday once it was successful.
But now there is no reaction at all. Not error, not success.
Serial says:
> getFile in downloadFile() [.pio/libdeps/esp32/FastBot2/src/FastBot2Client.h : 76]
> send + wait in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 268]
> got json in _parseResponse() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 308]
> got result in _parseResult() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 336]
> download file in downloadFile() [.pio/libdeps/esp32/FastBot2/src/FastBot2Client.h : 88]
> send + wait in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 268]
> got file in _parseResponse() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 323]
> ota write error in _ota() [.pio/libdeps/esp32/FastBot2/src/core/Fetcher.h : 76]
> send + wait in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 268]
> bad response in _parseResponse() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 330]
> getUpdates in getUpdates() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 209]
> send async in sendPacket() [.pio/libdeps/esp32/FastBot2/src/core/core.h : 271]

> in new update you can use
> 
> ```
> bot.updateFlash(u.message().document(), u.message().chat().id());
> ```
> 
> it's more stable

[image]

In case it has not been realised - will wait.

GyverLibs, 2024-08-30:
you need up to date version, 1.0.5

GyverLibs, 2024-08-30:
just tested on esp32 and C3 - works like charm

OleksandrGora, 2024-08-30:
Another issue while installation 1.0.5:
PIO Core Call Error: "Resolving esp32 dependencies...\nLibrary Manager: Installing gyverlibs/FastBot2 @ ^1.0.5\n\n\nUnknownPackageError: Could not find the package with 'gyverlibs/FastBot2 @ ^1.0.5' requirements for your system 'darwin_x86_64'"

Can't help? There is no issue with 1.0.3...

GyverLibs, 2024-08-30:
Just install from git. PIO doesn't update it's registry yet

OleksandrGora, 2024-09-03:
Tested for a few days. It seems it works. Thank you. Closing the issue.

closed 2024-09-03 by OleksandrGora

GyverLibs, 2024-09-03:
ncie

OleksandrGora, 2024-09-05:
Hello. Something the same:
[image]

GyverLibs, 2024-09-05:
Maybe firmware file is too big for partition? Try change partition or use atomic update with .gz compression