GyverLibs/FastBot2 #14: SSL: Unknown error code 0x4C

LittleBuster, 2024-09-07:
ESP32S3 N16R8
Framework ver.: 2.0.17

Заметил, что во время работы бота раз в 3 минуты в сериал SSL клиент печатает сообщение о неизвестной ошибке

```
[36217174][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[36427928][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[36638264][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[36848603][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[37058941][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[37269379][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[79424049][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[79635205][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[79845543][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[80055883][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[80266220][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
```

Basoil, 2024-11-30:
на ESP32-S3 N16 
Framework 3.0.7

регулярное в простое выдает:

```

2024-11-30 17:50:46 | [15564212][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
2024-11-30 18:18:40 | [17238947][E][ssl_client.cpp:36] _handle_error(): [data_to_read():401]: (-76) UNKNOWN ERROR CODE (004C)
2024-11-30 18:18:41 | [17238957][E][NetworkClientSecure.cpp:309] available(): Closing connection on failed available check

```

thedemoncat, 2025-05-14:
Подтверждаю, ошибка есть

сперва так :

`[587332][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)`

а затем так:

`[592689][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!`

сразу после нее всегда падает с исключением

`
0x4037d3af: _xt_context_save at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_context.S:194
0x42085b92: mbedtls_sha256_finish_ret at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/port/sha/dma/esp_sha256.c:254
0x4037d3af: _xt_context_save at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_context.S:194
0x4037407d: _xt_alloca_exc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1806
0x4037e1b9: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1554

`

GyverLibs, 2025-05-14:
Я читал что для 32 ещё не изобрели стабильную версию ssl клиента

GyverLibs, 2025-05-14:
Режима с отпечатком нет, не дошли руки сделать. Также читал что не рекомендуется разбивать работу с ссл клиентом на таски - оно обязательно посыпется. Особенно если учесть что в fb2 используется один клиент, он общий на чтение и отправку. Если разделить на таски - гарантировано будут проблемы

thedemoncat, 2025-05-14:
Тогда мне лучше все делать во одном методе? создать клиента, отправить сообщение и удалить?

GyverLibs, 2025-05-14:
Да, можно одного "бота" оставить глобально чтобы он пуллил апдейты, а сообщения отправлять с локальных ботов с флагом wait. Первое подключение к серверу занимает время, но в рамках другого таска это наверное неважно

thedemoncat, 2025-05-14:
Спасибо большое, попробую