GyverLibs/FastBot2 #16: Библиотека не работает с TinyGsm

Shatodor, 2024-09-23:
Использовал код из примера для ESP32 c модемом SIM7000, но с ним  тестовый пример minimal не работает. Через Wi-FI все ок.
В чем может быть причина?

```
// версия с другим клиентом, например tinygsm
#define TINY_GSM_MODEM_SIM7000
#include <TinyGsmClient.h>
TinyGsm modem(Serial);
TinyGsmClient gsmclient(modem);
#include <FastBot2Client.h>
FastBot2Client` bot(gsmclient);
```

GyverLibs, 2024-09-23:
Не знаю, внешний клиент не тестировался

Shatodor, 2024-09-23:
В какую сторону копать посоветуете?

GyverLibs, 2024-09-23:
Протестировать отдельно работу gsm, поделать запросы на сайты

Shatodor, 2024-09-30:
Нужно было использовать:

`#define TINY_GSM_MODEM_SIM7000SSL`

GyverLibs, 2024-09-30:
Ну, хорошо. Там вроде документация обширная, вся инфа есть

LittleBuster, 2024-10-02:
> Нужно было использовать:
> 
> `#define TINY_GSM_MODEM_SIM7000SSL`

После этого нормально заработало? А какой оператор?
Пробовал на Билайне на SIM900L там ужасно медленная скорость передачи данных через GPRS, даже обычные HTTP заголовки по полчаса грузило

Shatodor, 2024-10-02:
> > Нужно было использовать:
> > `#define TINY_GSM_MODEM_SIM7000SSL`
> 
> После этого нормально заработало? А какой оператор? Пробовал на Билайне на SIM900L там ужасно медленная скорость передачи данных через GPRS, даже обычные HTTP заголовки по полчаса грузило

Да, нормально. МТС

GyverLibs, 2024-10-02:
добавить в доку/примеры? Больше изменений в работе с tinygsm нету?

Shatodor, 2024-10-02:
> добавить в доку/примеры? Больше изменений в работе с tinygsm нету?

Было бы полезно указать, что для работы Телеграмма требуется защищенное соединение. Через Wi-Fi оно по умолчанию работает, а через GPRS нет.

GyverLibs, 2024-10-02:
что нужно для того чтобы оно работало, что указать в доке?

Shatodor, 2024-10-02:
Чтобы заработал FastBot2 для моего модема было достаточно  ` TINY_GSM_MODEM_SIM7000` заменить на ` TINY_GSM_MODEM_SIM7000SSL`. Как для других модемов не знаю.
Возможно даст наводку, то что у меня получилось отправлять HTTPS запросы через GPRS используя: 
```
TinyGsmClientSecure client(modem);
HttpClient http(client, server, port);
```

closed 2024-10-06 by GyverLibs