GyverLibs/FastBot2 #17: не видит файлы

AlexJust139, 2024-09-28:
Добрый день!

Версия 1.07, пробовал tick вставлять и в задаче Freertos, и в цикле Loop.
В любом случае - простые сообщения видит, а сообщения с приложенными файлами - игнорирует:
void updateh(fb::Update& u)   {     // ***************************** Telegram, вызывается из таска TGCheck *********************************

if (  u.message().hasDocument()      ) {
         Serial.println (" file downloaded");
     }

  if (u.isMessage()) { // именно сообщение
   fb::Message msg;
   if ( u.message().hasDocument()      )    {Serial.println ("OTA file!"); }
  String  text = u.message().text().decodeUnicode();
   int chatID = u.message().chat().id();
   int mes_id = u.message().id();
   Serial.print(text); Serial.print(" , chatID=");  Serial.print(chatID); Serial.print(", mes_id="); Serial.print(mes_id); 
  } // конец "именно сообщение"
} // finish handle telegram message

После включения #define FB_USE_LOG Serial вижу следующее:
got json in _parseResponse() [C:\arduino\VSC\libraries\FastBot2\src/core/core.h : 409]
> getUpdates in getUpdates() [C:\arduino\VSC\libraries\FastBot2\src/core/core.h : 264]
> send async in sendPacket() [C:\arduino\VSC\libraries\FastBot2\src/core/core.h : 326]
В чем может быть дело?

GyverLibs, 2024-09-28:
привет, проверил - всё работает, esp8266/esp32
```cpp

void updateh(fb::Update& u) {
    Serial.println("NEW MESSAGE");

    if (u.isMessage()) {
        Serial.println("message!");
        Serial.println(u.message().text());
    }
    if (u.message().hasDocument()) {
        Serial.println("doc!");
        // Serial.println(u.message().caption());
        Serial.println(u.message().document().name());
    }
}
```

AlexJust139, 2024-09-29:
Алекс, спасибо большое!
Пошел копать в сторону самого чата:
- проверил бота на параметр Group Privacy - все ок.
- затем наткнулся на то, боту надо быть админом чата - он им не был.
Сделал админом - все сразу заработало!
Тикет можно закрывать, а про админа имхо имеет смыл добавить в доки.

AlexJust139, 2024-09-29:
There were incorrect chat setting (not a Fastbot2 issue)

closed 2024-09-29 by AlexJust139

GyverLibs, 2024-09-29:
Каиф

GyverLibs, 2024-09-29:
Про сделать админом в доке написано в самом начале

AlexJust139, 2024-09-29:
видимо, плохо искал. Хотя смотрел и в README.md и в /docs/1.main.md и в main/docs/7.files.md (

GyverLibs, 2024-09-29:
https://github.com/GyverLibs/FastBot2/blob/main/docs/2.bot.md

[image]

AlexJust139, 2024-09-29:
посыпаю голову пеплом...
давно пробегался по этой странице и потом не возвращался.
Алекс, спасибо!

GyverLibs, 2024-09-29:
не очень очевидно да, но всю информацию без дублирования в одной доке уместить непросто)