GyverLibs/FastBot2 #18: Ошибка парсинга многосоставной HTML строки

LittleBuster, 2024-10-02:
Версия бота 1.0.8
ESP32S3 N16R8
Framework ver.: 2.0.17

Пример отправки сообщения
```
    fb::Message resp;
    fb::Menu    menu;

    resp.chatID = user->chatId;
    resp.mode = fb::Message::Mode::HTML;

    String name = "test";
    resp.text = "<b>" + name + "</b>";
   //resp.text = String("<b>") + String("test") + String("</b>");

    resp.setMenu(menu);
    bot.sendMessage(resp);
```
В core.h в функции _parseResponse функция res.parseJson(); возвращает false.
В result.h с ошибкой выходит на этой строке
```
if (!_parser[tg_apih::ok].toBool()) return 0;
```
Решается сложением строк обходным путем:
```
    resp.text.concat("<b>");
    resp.text.concat(ctrl->getName());
    resp.text.concat("</b>");
```

GyverLibs, 2024-10-02:
видимо какие то проблемы в ардуино ядре, хорошо бы вывести полученную строку в порт почитать

[image]

LittleBuster, 2024-10-02:
Какая-то магия. В обоих случаях строки одинаковые.
```
    for (size_t i = 0; i <= resp.text.length(); i++) {
        Serial.printf("[%u] = %c = %u\n", i, resp.text.c_str()[i], resp.text.c_str()[i]);
    }
```
```
[0] = < = 60
[1] = b = 98
[2] = > = 62
[3] = t = 116
[4] = e = 101
[5] = s = 115
[6] = t = 116
[7] = < = 60
[8] = / = 47
[9] = b = 98
[10] = > = 62
[11] = ␀ = 0
```
```
[0] = < = 60
[1] = b = 98
[2] = > = 62
[3] = t = 116
[4] = e = 101
[5] = s = 115
[6] = t = 116
[7] = < = 60
[8] = / = 47
[9] = b = 98
[10] = > = 62
[11] = ␀ = 0
```

GyverLibs, 2024-10-02:
Определенно магия

LittleBuster, 2024-10-02:
Возможно где-то что-то кораптится в памяти в проекте большем чем просто HelloWorld, со множеством библиотек и кода. Проверил на чистом проекте на примере minimal таких проблем нет.
Делал FullClean в VScode - ничего не дает.

GyverLibs, 2024-10-02:
Ох такое сложно поймать будет