GyverLibs/FastBot2 #20: Proxi

impulseDiMaro, 2024-10-15:
День добрый
вопрос должна ли работать библиотека через прокси?
в свете последних событий в некоторых областях страны многие остались без сервиса
для десктопных и мобильных версий помогает прокси или впн
как быть с микроконтроллером?

// установить proxy
void setProxy(const char* host, uint16_t port);
// установить proxy
void setProxy(const IPAddress& ip, uint16_t port);

попробовал... глухо пока. протоколы HTTP Socks5
может зря пытаюсь? версия тестовая, та еще, что без зависимости от GyverHTTP

GyverLibs, 2024-10-17:
> нашел не очень быстро

все зависимости ставятся автоматически, не нужно ничего искать)

impulseDiMaro, 2024-10-17:
ну это у Вас, на Visual Studio c Platformio... на IDE такого же нет... я так думаю  :)

GyverLibs, 2024-10-17:
ide сама качает все зависимости

impulseDiMaro, 2024-10-17:
мне это не известно... бегло прошел по настройкам, в недоумении

impulseDiMaro, 2024-10-17:
попробовал платный прокси HTTPS с авторизацией по IP. так же... не летят пакеты с бота, с мессенджера нормально

impulseDiMaro, 2024-10-20:
добавил в вебку установку прокси, о которой, до блокировке даже не думал... надоело вручную перебирать хосты и порты

[image]

увеличил ожидание ответа сервера с 2 сек до 10
нет коннекта с сервером

лог без прокси:

>  getUpdates in getUpdates() [src/core/core.h : 278]
> [String] 'POST https ...       

> ': Reallocating large String(207 -> 208 bytes)
> [String] 'POST https ... eout":20,"': Reallocating large String(222 -> 227 bytes)
> [String] 'POST https ... 3,"offset"': Reallocating large String(239 -> 240 bytes)
> [hostByName] request IP for: api.telegram.org
> [hostByName] Host: api.telegram.org IP: 149.154.167.220
> BSSL:connect: Unable to connect TCP socket
> send error in sendPacket() [src/core/core.h : 350]

с прокси HTTP:

> getUpdates in getUpdates() [src/core/core.h : 278]
> [String] 'POST https ...       

> ': Reallocating large String(207 -> 208 bytes)
> [String] 'POST https ... eout":20,"': Reallocating large String(222 -> 227 bytes)
> [String] 'POST https ... 3,"offset"': Reallocating large String(239 -> 240 bytes)
> [hostByName] Host: 185.149.21.186 is IP!
> BSSL:_connectSSL: start connection
> BSSL:_wait_for_handshake: failed
> BSSL:Couldn't connect. Error = 'Incoming protocol or record version is unsupported.'
> send error in sendPacket() [src/core/core.h : 350]

impulseDiMaro, 2024-10-27:
в общем покурив тему, пришел к выводу что толку от прокси в моем случае мало, так как нет шифрования, ESP это "не по зубам" и все запросы перехватывается по пути следования и... "от ворот поворот"
нужен посредник который сможет зашифровать.
на свободном роутере с OpenWRT на котором крутиться только локальный MQTT сервер и локальный NTP сервер, поднял VPN клиента а на удаленном (арендованный) VPN сервер.
на главном роутере хосты телеграмма заворачиваются на роутер с поднятым туннелем на VPN сервер, остальное летит как обычно.
таким образом удалось восстановить работу ботов.