GyverLibs/FastBot2 #21: Проблема с MarkdownV2 при редактировании сообщения

leowerd, 2024-10-16:
Создание сообщения с разметкой MarkdownV2 и HTML работает и отображается корректно, но при редактировании уже отправленного сообщения работает только разметка HTML, при выборе MarkdownV2 сообщение не изменяется.

Так работает и отправляет:
```
    fb::Message msgInfo("`Инфо`", CHAT_ID);
    msgInfo.mode = fb::Message::Mode::MarkdownV2;
    bot.sendMessage(msgInfo);
    infoID = bot.lastBotMessage();  // запомнили ID сообщения с инфо

```
Так работает и меняет текст:
```
    String infoMessage = "Станция: " + String(lastStation) + "\n" + String(lastTitle) + "\n<code>Громкость: " + String(lastVolume) + "  Битрейт: " + String(lastBitrate) + " kbps</code>\n";
            fb::TextEdit editInfo;
            editInfo.mode = fb::Message::Mode::HTML;
            editInfo.text = infoMessage;
            editInfo.chatID = CHAT_ID;
            editInfo.messageID = infoID;
            bot.editText(editInfo);

```
Так не работает и текст сообщения остается старым:
```
    String infoMessage = "Станция: " + String(lastStation) + "\n" + String(lastTitle) + "\n`Громкость: " + String(lastVolume) + "  Битрейт: " + String(lastBitrate) + " kbps`\n";
            fb::TextEdit editInfo;
            editInfo.mode = fb::Message::Mode::MarkdownV2;
            editInfo.text = infoMessage;
            editInfo.chatID = CHAT_ID;
            editInfo.messageID = infoID;
            bot.editText(editInfo);
```
Пробовал экранировать через \ но это ничего не дало, другие теги MarkdownV2 тоже пробовал.

В остальном бот работает хорошо.
esp32 wroom, arduino 1.8.19, FastBot2-1.0.10

GyverLibs, 2024-10-16:
Возможно ограничение телеграм

GyverLibs, 2024-10-16:
а не, работает всё. Внимательнее к сложению строк, оно выглядит максимально коряво в коде выше

[image]

leowerd, 2024-10-16:
Сейчас перепроверил
В таком виде работает обновление:

[image]

А в таком виде уже нет:

[image]

Вроде в строке ничего криминального не осталось :)

Просто остается старый текст и не меняется:

[image]

Дело в том, что на первой версии бота MarkdownV2 работал хорошо, и тот кусок отлично отображался, как то так:

[image]

Проблема пришла с переходом на fastbot2, я сейчас переделал на HTML, но решил сказать об этом, вдруг это какой-то баг

GyverLibs, 2024-10-16:
Я сегодня обновил, добавил обработчик ошибок. Если текст не меняется - скорее всего телеграм отвечает ошибкой, и будет полезно её описание почитать. Хотя на предыдущей версии можно просто вывести так
```
fb::Result res = bot.editText...
Serial.println(res.getRaw());
```

GyverLibs, 2024-10-16:
Возможно это какой то глубинный баг со строками, в соседнем issue наблюдается похожее колдунство и пока что его отладить не удалось

GyverLibs, 2024-10-16:
не могу повторить баг. Нужен конкретно прям код где это наблюдается =(
[image]

[image]

leowerd, 2024-10-16:
> Я сегодня обновил, добавил обработчик ошибок. Если текст не меняется - скорее всего телеграм отвечает ошибкой, и будет полезно её описание почитать. Хотя на предыдущей версии можно просто вывести так
> 
> ```
> fb::Result res = bot.editText...
> Serial.println(res.getRaw());
> ```

Что интересно, в консоль ничего не выводит

[image]

[image]

leowerd, 2024-10-16:
> не могу повторить баг. Нужен конкретно прям код где это наблюдается =( 
Попробую как-нибудь воспроизвести баг на меньшем куске кода, чтобы передать потом)

GyverLibs, 2024-10-16:
можно в конфиге bot_config.h раздефайнить отладку и глянуть монитор
```
// #define FB_USE_LOG Serial
```

impulseDiMaro, 2024-10-17:
дык в строке символ "-" заэкранировать надо и должно заработать, нет?

GyverLibs, 2024-10-17:
кстати да
`Bad Request: can't parse entities: Character '-' is reserved and must be escaped with the preceding '\\'`

leowerd, 2024-10-17:
> дык в строке символ "-" заэкранировать надо и должно заработать, нет?

Вы правы, сейчас проверил, действительно проблема в этом.
Странно что на первой версии все работало нормально.

> кстати да

Получается не баг, а фича, придется долонительную обработку ставить, тк получение этих символов от сервера неконтролируемо, в режиме экономия ресурсов выглядит как лишний рот))

Это на стороне телеграмма ограничение, полагаю, судя по описанию апи, хотя символ не используется в разметке, странно с их стороны, учитывая правило:

``In all other places characters '_', '*', '[', ']', '(', ')', '~', '`', '>', '#', '+', '-', '=', '|', '{', '}', '.', '!' must be escaped with the preceding character '\'.``

И то, что в их же примерах эти символы не экранированы. 

[image]

Да, кстати, если раздефайнить отладку, то показывает это:

[image]

В любом случае, спасибо за беспокойство :)

impulseDiMaro, 2024-10-17:
> И то, что в их же примерах эти символы не экранированы. 

полагаю что это не примеры а описание разметки
ибо нет смысла в форматированный блочный код запихивать строку типа "pre-formaеted..."
типа такого, да
[image]
удачи Вам в доработке youradio, хороший, грамотный проект

GyverLibs, 2024-10-17:
суть в том, что экранировать нужно то, что должно выводиться как обычный текст. В первом фастботе встроено экранирование для некоторых символов, но из за этого он тоже не очень корректно работает. Я добавил инфу в доку. По сути это косяк телеграма, могли бы парсить markdown нормально, а не отдавать парсинг на юзера

leowerd, 2024-10-18:
> удачи Вам в доработке youradio, хороший, грамотный проект

Спасибо) Как доведу плагин для радио до ума, выложу код на форуме Алекса

> Я добавил инфу в доку. По сути это косяк телеграма, могли бы парсить markdown нормально, а не отдавать парсинг на юзера

Благодарю, думаю это многим пригодится :)
Пс, библиотека огонь)

GyverLibs, 2024-10-18:
Я попробую встроить умную экранизацию, чтобы она не ломала md конструкции. Но на все символы наверное не получится все равно