GyverLibs/FastBot2 #22: Проблема с InlineMenu

SCratORS, 2024-10-17:
ESP32, PlatformIO.
В обработчик Update не приходят коллбеки от кнопок InlineMenu.
Даже пробовать прошить [inlineMenu.ino](https://github.com/GyverLibs/FastBot2/blob/main/examples/inlineMenu/inlineMenu.ino)
Пробовал по всякому. Коллбеки тупо не приходят, вообще никакой реакции на кнопки. Даже в логах в Serial ничего не пишет. Вообще ничего.

GyverLibs, 2024-10-17:
хз, всё работает, примеры тоже

GyverLibs, 2024-10-17:
можно в конфиге bot_config.h раздефайнить отладку и глянуть монитор
```
// #define FB_USE_LOG Serial
```

SCratORS, 2024-10-17:
Делал. Ничего не даёт. идут обычные логи. никакой полезной информации или хоть какой-то реакции на кнопки inlineMenu там не прилетает.

GyverLibs, 2024-10-17:
значит кнопка не нажимается) надо нажимать лучше

SCratORS, 2024-10-17:
Прикол засчитан =D,.. но анимация нажатия и ожидания коллбека на кнопке появляется =)

closed 2024-10-18 by SCratORS

GyverLibs, 2024-10-18:
Во, яж говорил

SCratORS, 2024-10-18:
А если серьёзно, то почему-то после того, как я через BotFather добавил в бота команды, бот перестал слать коллбеки на кнопки InlineMenu. Удалил команды, - не помогло. 
Помогло удаление бота, и создание нового.

GyverLibs, 2024-10-18:
Может ещё какие нибудь штуки типа приватности и прочих настроек слетели

SCratORS, 2024-10-18:
А вот теперь вопрос:
1. Создал нового бота.
2. Написал прошивку на ESP32 для работы с ботом, не используя никаких библиотек ботов, тупо на HTTPClient.h
3. Бот отправляет inline_keyboard, коллбеки от ее приходят, всё работает.
4. Загружаю прошивку из примера FastBot2. Коллбеки от inline_keyboard перестают приходить.
5. Загружаю прошивку из п.2 - Аналогично, коллбеки тоже теперь не приходят.
--- Я не знаю что это и с чем кушать.

GyverLibs, 2024-10-18:
Ну это уже эзотерика...

GyverLibs, 2024-10-18:
Самый стопудовый вариант проверить бота - вручную в браузере запросить getupdates

SCratORS, 2024-10-18:
getupdates возвращается пустой
"ok":true
"result": []
Ладно, сам понимаю что какая-то чушь. Но вот так.

woodlist, 2024-10-21:
> Вот действительно хз.. Сделал чистую установку ArduinoIDE, чистую ESP32 Поставил в ArduinoIDE либы: GTL-1.1.7 GyverHTTP-1.0.21 StringUtils-1.4.26 GSON-1.5.10 FastBot2-1.0.11 Взял пример [inlineMenu.ino](https://github.com/GyverLibs/FastBot2/blob/main/examples/inlineMenu/inlineMenu.ino) Вписал в Define свои данные. Залил в ESP32 через программатор и фиг. InlineMenu в чат прилетает, но при нажатии на кнопку - тишина. Нигде никакой реакции. Пробовал создавать InlineMenu через AddButton - Тоже самое. Пробовал не прописывать "data" - тоже самое. Пробовал менять setPollMode - тоже самое.
> 
> Изменил update на `void updateh(fb::Update& u) { Serial.println("UPDATE"); }` Нифига. Текст если отправить - пишет в Serial, а от InlineMenu - Никакой реакции.
> 
> Может можно как-то где-то еще отловить коллбек кроме как в "update"? При этом FastBot.h - работает нормально, но медленнее.

JSON библиотеки разные бывают. От Benoit Blanchon версии 7.2.0 вполне совместимый.

GyverLibs, 2024-10-21:
@woodlist это к чему? Совместимый с чем?

woodlist, 2024-10-27:
> @woodlist это к чему? Совместимый с чем?

Я малость напутал, GSON с JSON. Так что отзываю свою прежнюю запись.

GyverLibs, 2024-10-27:
Gson это библиотека, а json это формат данных

DymitriPowell, 2025-03-14:
У меня такаже проблема. Нет коллбеков от кнопок Инлайн-меню. Lolin Wemos esp8266. Arduino IDE 2.3.4. Все описанные выше манипуляции проводил. Только нового бота не создавал еще. Результата нет.

SCratORS, 2025-03-14:
> Вывод - какие то проблемы с настройками самого бота

Если бы проблема была в настройках бота, то инлайн-меню не работало бы на любых либах, а не только на данной.
Лично мое мнение, что в этой либе из-за очень сильной оптимизации, где-то что-то порезали, или не добавили, толи в модуле парсинга JSON, толи в HTTP клиенте. То есть JSON отправляется толи с ошибкой синтаксиса, толи какой-то байт типа конца строки или перевода каретки добавляется где не надо или не добавляется где надо. И в итоге сам телеграмм начинает не корректно отрабатывается. Я хз. Вывод я этот делаю из того, что до использования этой либы инлайн-меню работают, а после интайн-меню перестают работать даже если взять другие либы или запросы формировать руками.

DymitriPowell, 2025-03-14:
А как тогда объяснить то, что к меня сейчас два бота. Старый и новый. Я в скетче меняю только токен бота и скетч на старом боте не работает, а на новом работает. Один и тот же скетч, одни и те же библиотеки. 
Может есть какой то кэш в телеге? Может какие-то мои предыдущие действия со первым ботом не дают нормально дальше с ним работать?

SCratORS, 2025-03-14:
Может в версии 1.0.12 пофиксили.... 
у меня нет ответа. сам баг очень странный и пахнет бредом. Точно, что если они перестают работать, то отваливается на строне самой телеги. Но точно тут не проблемы в настройках приватности или еще чего. Кеш... врядли. Если конечно он не в БД телеги, или типа того.
Так или иначе, мне пришлось отказаться от этой либы - и жаль и не очень жаль.
Жаль - т.к. довольно простое и быстрое решение.
Не жаль - я написал все запросы и их обработку вручную, что позволило сделать всё более "профильно", и смог еще сильнее поднять скорость обработки, т.к. добавил 2 бот-клиента, один висит в лонг-пуле и работает на внешние события (команды), второй отправляет сообщения по внутренним событиям (детекторы, сенсоры). Получилось мега быстрая работа бота.

GyverLibs, 2025-03-20:
В целях оптимизации тут ничего не порезано и не уменьшено, всё полноценное и json и http