GyverLibs/FastBot2 #23: Просьба в дополнении примеров

woodlist, 2024-10-21:
Пожалуйста, добавьте Replykeyboard пример.

GyverLibs, 2024-10-21:
Есть в доке и примере test

closed 2024-10-21 by GyverLibs

woodlist, 2024-10-21:
Да, действительно запрашиваемый пример был в menu. Путаница у меня возникла из за терминологии. У всех других поставщиков библиотек оно называется Reply Keyboard, у Вас называется "Большая клавиратура".

woodlist, 2024-10-21:
С удовлетворением хочу отметить, что Ваша библиотека единственная рабочая среди прочих схожих после обновлении Ardunio IDE до последней версии и JSON.

GyverLibs, 2024-10-21:
Потому что я использую свой стек библиотек и не завишу от чьих то обновлений :)

woodlist, 2024-10-21:
Маэстро, раз так получилось, что мы ведем беседу на закрытой ветке, не отвлекая других, задам такой вопрос.
`fb::InlineMenu menu("kek 1 ; kek 2 ; kek 3 \n kek 4 ; kek 5", "test;pest;lol;https://www.google.ru/;https://www.yandex.ru/");`
имеет аналог для `Replykeyboard?`
Я попытался по хешу обработать нажатие на кнопки, но не получилось.

GyverLibs, 2024-10-21:
Reply keyboard имеет только текст на кнопке, его же отправляет при нажатии просто в чат

woodlist, 2024-10-21:
Как жаль... А это будет трудно наделить его таким желанным свойством?

GyverLibs, 2024-10-21:
Думаю да. Нужно написать Пашке Дурову и убедить его, что у reply keyboard не хватает коллбэков

woodlist, 2024-10-21:
Ах этот проказник Павел виноват? Тогда снимаю вопрос.

woodlist, 2024-10-21:
Шутка шуткой, но просветите пожалуйста. Получается нажатие Reply кнопки вызывает действие только в пределах окна бота и ESP32 ни принимает и не сможет обработать такое нажатие?

GyverLibs, 2024-10-21:
Нажатие reply keyboard просто отправляет текст с кнопки в чат с ботом, этот текст обрабатывается как обычное сообщение

GyverLibs, 2024-10-21:
Об этом написано в документации в первом же абзаце

https://github.com/GyverLibs/FastBot2/blob/main/docs/6.menu.md

woodlist, 2024-10-23:
Спасибо большое. У меня все получилось. И код загрузки удивительно небольшое и скорость взаимодействия с Телеграмом впечатляющий. Маэстро, как можно начислять Вам деньги на кружку ядреного пива, с рыбкой на закуску? Пожалуйста, только международный способ.

GyverLibs, 2024-10-23:
Только то что по ссылке в шапке репозитория, международных наверное нету. Хотя boosty может принимает

woodlist, 2024-10-24:
Маэстро, проверьте Бусти. Кажется получился.

GyverLibs, 2024-10-24:
Спасибо, получил