GyverLibs/FastBot2 #24: Отправка файла с СД карты

JesseOceanMusic, 2024-10-31:
Во-первых хотел бы сказать спасибо... Пару месяцев сидел на Universal Telegram - замучился её пытаться дописывать, перешёл на FastBot2 и скорость, конечно, радует) кудесник, не иначе)

```
void LOGread()                                   // чтение лога //
{
  if (!SD.begin(4))
  {
    send_alert("ERROR: Ошибка инициализации SD карты");
  }
  
  String file_name = object_TimeDate.get_DateFULL() + ".txt";
  File myFile = SD.open(file_name);

  if (myFile)
  {
    fb::File f(file_name, fb::File::Type::document, myFile);
    f.chatID = object_array_users[0].get_id();
    bot_main.sendFile(f);
  }

  else
  {
    send_alert("ERROR: Ошибка чтения SD карты");
  }

  myFile.close();
}
```

Собственно, вопрос - чего я неправильно делаю? Обычный .txt файл. В `Serial` у меня его выводить получается, с помощью Universal его отправлял, а с FastBot2 не получается...

Спасибо!

GyverLibs, 2024-10-31:
Пример из библиотеки с отправкой работает? Id чата точно корректный? Можно включить вывод логов в bot config файле, также можно подключить обработку ошибки (см доку) и вывести в сериал, может телеграм что то отвечает

JesseOceanMusic, 2024-10-31:
Уже начал отвечать, как 100500 ребилд заработал)) На форуме [тоже писал](https://community.alexgyver.ru/threads/fastbot-obsuzhdenie-biblioteki.6896/post-176264), щас туда тоже отпишусь!)

Это не баг а фича, я не понимаю почему так)

Прежде, чем я дошёл до этого - я перепробовал всё, что только в голову пришло... В том числе начал менять настройки Async/Sync/Long, таймауты, лимиты... Дилеи пытался добавлять итд итп. Даже бота нового создал, на всякий случай...

**Решение:**
_Перепроверил два раза меняя только эти полторы строки._

_вот так не работает!_
```
f.chatID = object_array_users[users_array_index].get_id();    // метод класса возвращающий стринг с айди пользователя
```
_а вот так работает!_

```
String buf_user_ID = object_array_users[users_array_index].get_id();    // метод класса возвращающий стринг с айди пользователя
f.chatID = buf_user_ID;
```

PS - еще раз спасибо за библиотеку!

PSS - может f.chatID хранит ссылку, а метод .get_id() передаёт Стринг по значению, а сам стринг лежит в приватной переменной класса... Пока это моя единственная догадка.

GyverLibs, 2024-10-31:
Что есть `f`?

JesseOceanMusic, 2024-10-31:
`fb::File f(file_name, fb::File::Type::document, myFile);`

GyverLibs, 2024-10-31:
Да, chatID хранит указатель на строку. Возможно стоит заменить его на стрингу, чтобы не было подобных проблем

JesseOceanMusic, 2024-10-31:
Спасибо!) доброй ночи!))

closed 2024-11-02 by JesseOceanMusic