GyverLibs/FastBot2 #26: Отправка из буфера ESP32-CAM

KudPavel, 2024-11-30:
Пытаюсь отправить фото из буфера ESP32-CAM (fbcam,) в телеграмм почему то ничего не приходит.

```
 void sendPhotoTelegram() {
  //Dispose first picture because of bad quality
  camera_fb_t * fbcam = NULL;
  fbcam = esp_camera_fb_get();
  esp_camera_fb_return(fbcam); // dispose the buffered image
  // Take a new photo
  fbcam = NULL;  
  fbcam = esp_camera_fb_get(); 
  if(!fbcam) {
    Serial.println("Camera capture failed");
    delay(1000);
    ESP.restart();
  }  
    uint8_t *fbBuf = fbcam->buf;
    size_t fbLen = fbcam->len;
           Serial.println("отправляем снимок");
           //fb::File f("file.txt", fb::File::Type::document, "https://compote.slate.com/images/697b023b-64a5-49a0-8059-27b963453fb1.gif");
           fb::File f("esp32-cam.jpg", fb::File::Type::photo, fbBuf, fbLen); 
            f.chatID = chat_id;
            //bot.sendFile(f, false);    
             bot.sendFile(f, true);

    esp_camera_fb_return(fbcam);
}    
```

Если bot.sendFile(f, false) то в логах:
```
> send async in sendPacket() 
> poll timeout in tick() 
> getUpdates in getUpdates()
> send async in sendPacket()
> got json in _parseResponse()
```
Если bot.sendFile(f, true):

```
> send + wait in sendPacket()
> bad response in _parseResponse() 
> getUpdates in getUpdates() 
> send + wait in sendPacket()
> got json in _parseResponse() 
```

GyverLibs, 2024-11-30:
отправлять нужно обязательно синхронно (send (true)), после отправки делать fb_return() или как там, из стандартных примеров с камерой. Камера должна быть настроена на JPEG снимки

KudPavel, 2024-12-02:
Отправляю синхронно  bot.sendFile(f, true),  
esp_camera_fb_return(fbcam) делаю, 
config.pixel_format = PIXFORMAT_JPEG; 
Как уже только не пробовал, всё равно ничего не приходит.
Как я понял camera_fb_t * fbcam это структура, в которой содержится:
unit8_t * buf - указатель на данные снимка (Pointer to the pixel data)
size_t len - длина буфера в байтах
В принципе я их и отправляю в fb::File, что не так, непонятно.
    uint8_t *fbBuf = fbcam->buf;
    size_t fbLen = fbcam->len;

GyverLibs, 2024-12-02:
Отправляем вот так:

```cpp
camera_fb_t* fbj = esp_camera_fb_get();

fb::File f("frame.jpg", fb::File::Type::photo, fbj->buf, fbj->len);
f.chatID = u.message().chat().id();
bot.sendFile(f, true);

esp_camera_fb_return(fbj);
```

Полный код примера, отправка по команде `/start`:

```cpp
#include <Arduino.h>
#include <FastBot2.h>
#include <WiFi.h>

#include "camera.h"

#define WIFI_SSID ""
#define WIFI_PASS ""
#define BOT_TOKEN ""

FastBot2 bot;

void updateh(fb::Update& u) {
    Serial.println("New message!");

    if (u.message().text() == "/start") {
        camera_fb_t* fbj = esp_camera_fb_get();

        fb::File f("frame.jpg", fb::File::Type::photo, fbj->buf, fbj->len);
        f.chatID = u.message().chat().id();
        bot.sendFile(f, true);  // обязательно синхронная отправка

        esp_camera_fb_return(fbj);
    }
}

void setup() {
    Serial.begin(115200);
    Serial.println();

    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID, WIFI_PASS);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("Connected");
    Serial.println(WiFi.localIP());

    bool res = cam_init(FRAMESIZE_VGA, PIXFORMAT_JPEG);
    Serial.println(res ? "Camera OK" : "Camera ERROR");

    bot.attachUpdate(updateh);
    bot.setToken(F(BOT_TOKEN));
    bot.setPollMode(fb::Poll::Long, 20000);
}

void loop() {
    bot.tick();
}
```

KudPavel, 2024-12-02:
Ваш код мне нравится больше, но дело как оказалось не в этом.
Я установил FRAMESIZE_VGA как в вашем примере и мой код заработал, но если повысить разрешение до SVGA, то фото перестают отправляться.
Я делал на основе примера с  [Universal Telegram Bot Library](https://randomnerdtutorials.com/telegram-esp32-cam-photo-arduino/) который без проблем отправляет фото с такими настройками:

```
  config.pixel_format = PIXFORMAT_JPEG;
  config.grab_mode = CAMERA_GRAB_LATEST;

  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;  //0-63 lower number means higher quality
    config.fb_count = 1;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;  //0-63 lower number means higher quality
    config.fb_count = 1;
  }
```
И те же самые настройки прописаны у меня. Но мой код с использованием FastBot2 получается может отправить только VGA разрешение, как только я ставлю SVGA и выше, фото перестают отправляться.
А ваш пример не могу запустить, ругается на #include "camera.h" ( я использую "esp_camera.h") и на 'cam_init'.

GyverLibs, 2024-12-02:
понял принял, это косяк на стороне ядра есп, он не в состоянии отправить весь буфер целиком. Очень тупая фигня, придётся вызывать отправку кусками...

KudPavel, 2024-12-03:
Пример отлично работает. Спасибо. Правда компилятор ругается на повторную декларацию camera_fb_t* fbj = esp_camera_fb_get(), но это не проблема.
Но после обновления gyverhttp, мой код перестал работать, сейчас пробую переделать свой код в соответствие с вашим примером, я думаю должно всё получиться.

GyverLibs, 2024-12-03:
Ой да, поправлю

KudPavel, 2024-12-03:
Всё, супер, у меня всё работает.

closed 2024-12-03 by KudPavel