GyverLibs/FastBot2 #27: Проблема компиляции мин. примера

arrrtema1, 2024-12-01:
Загружаю мин. пример, библы все стоят, в чем может быть причина?

In file included from c:\Users\������������\Documents\Arduino\libraries\GyverHTTP\src/utils/Client.h:10:0,
                 from c:\Users\������������\Documents\Arduino\libraries\GyverHTTP\src/GyverHTTP.h:3,
                 from c:\Users\������������\Documents\Arduino\libraries\FastBot2\src/core/Fetcher.h:3,
                 from c:\Users\������������\Documents\Arduino\libraries\FastBot2\src/core/core.h:11,
                 from c:\Users\������������\Documents\Arduino\libraries\FastBot2\src/FastBot2Client.h:2,
                 from c:\Users\������������\Documents\Arduino\libraries\FastBot2\src/FastBot2.h:13,
                 from C:\Users\Пользователь\AppData\Local\Temp\.arduinoIDE-unsaved2024111-10664-9h1oao.z6bdf\inlineMenu\inlineMenu.ino:8:
c:\Users\������������\Documents\Arduino\libraries\GyverHTTP\src/StreamReader.h: In constructor 'StreamReader::StreamReader(Stream*, size_t, bool)':
c:\Users\������������\Documents\Arduino\libraries\GyverHTTP\src/StreamReader.h:37:153: error: 'class Stream' has no member named 'getTimeout'
     StreamReader(Stream* stream = nullptr, size_t len = 0, bool chunked = false) : stream(stream), _len(len), _chunked(chunked), _tout(stream ? stream->getTimeout() : READER_DEF_TOUT) {}
                                                                                                                                                         ^

exit status 1

Compilation error: exit status 1

GyverLibs, 2024-12-01:
Видимо какая то старая или корявая версия ядра esp

arrrtema1, 2024-12-01:
2.7.4 -> 3.1.2 все работает, спасибо

closed 2024-12-01 by arrrtema1