GyverLibs/FastBot2 #28: ограничение bot.downloadFile 227 байт?

LesnikEvg, 2024-12-05:
На ESP8266 при попытке загрузить в чат и сохранить на карте памяти текстовый файл, наткнулся на ограничение 227 байт. Есть такое? Возможно ли исправить? Короткий пример:
/*

*/

```
#define WIFI_SSID ""
#define WIFI_PASS ""
#define BOT_TOKEN ""
#define CHAT_ID ""

#include <FastBot.h>
FastBot bot(BOT_TOKEN);

#include <SD.h> 

void setup() {
  connectWiFi();
  SD.begin(15);
  bot.attach(newMsg);
}

// обработчик сообщений
void newMsg(FB_msg& msg) {
  // выводим всю информацию о сообщении
  Serial.println(msg.toString());

  if (msg.isFile) {                     // это файл
    Serial.print("Downloading ");
    Serial.println(msg.fileName);

    String path = '/' + msg.fileName;   // вида /filename.xxx
    File f = SD.open(msg.fileName, FILE_WRITE);  // открываем для записи
    bool status = bot.downloadFile(f, msg.fileUrl);
    Serial.println(status ? "OK" : "Error");

    f = SD.open(msg.fileName);
    bot.sendMessage(msg.fileName, CHAT_ID);
    Serial.println(bot.sendFile(f, FB_DOC, msg.fileName, CHAT_ID) ? "FOK" : "FError"); //FB_PHOTO
    f.close();
  }
}

void loop() {
  bot.tick();
}

void connectWiFi() {
  delay(2000);
  Serial.begin(115200);
  Serial.println();

  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (millis() > 15000) ESP.restart();
  }
  Serial.println("Connected");
}
```

GyverLibs, 2024-12-05:
нет, никаких ограничений нет. А на littlefs получается сохранить?

LesnikEvg, 2024-12-06:
На LittleFS получается сохранить до 2 мб текста .txt. При этом картинка в формате .bmp, сохранятеся только 221-229 байт. На карту не сохранить ни текст ни картинку больше этих байт. Извиняюсь, не в то обсуждение закинул, использую первый фастбот..
```
/*

*/

#define WIFI_SSID ""
#define WIFI_PASS ""
#define BOT_TOKEN ""
#define CHAT_ID ""

#include <FastBot.h>
FastBot bot(BOT_TOKEN);

//#include <LittleFS.h>
#include <SD.h> 

void setup() {
  connectWiFi();
  //LittleFS.begin();
  SD.begin(15);
  bot.attach(newMsg);
  bot.sendMessage("Здравствуйте", CHAT_ID);
}

// обработчик сообщений
void newMsg(FB_msg& msg) {
  // выводим всю информацию о сообщении
  //Serial.println(msg.toString());

  if (msg.isFile) {                     // это файл
    Serial.print("Downloading "); Serial.println(msg.fileName);

    String path = msg.fileName;   //  '/' + 

    //File f = LittleFS.open(path, "w");
    File f = SD.open(path, FILE_WRITE);  // открываем для записи
    int status = bot.downloadFile(f, msg.fileUrl);
    Serial.print(status);
    Serial.println(status ? "Скачал" : "Не скачать");
    f.close();

    //f = LittleFS.open(path, "r");
    f = SD.open(path);
    bot.sendMessage(path, CHAT_ID);
    int status2 = bot.sendFile(f, FB_PHOTO, path, CHAT_ID);// FB_DOC
    Serial.print(status2);
    Serial.println(status2 ? "Отправил" : "Не отправил"); 
    f.close();
  }
}

void loop() {
  bot.tick();
}

void connectWiFi() {
  delay(2000);
  Serial.begin(115200);
  Serial.println();

  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (millis() > 15000) ESP.restart();
  }
  Serial.println("Connected");
}

```

GyverLibs, 2024-12-06:
В первом без понятия, там стандартные библиотеки используются