GyverLibs/FastBot2 #29: Не работает OTA

vvzvlad, 2024-12-28:
А какие настройки нужны для OTA? В первой версии библиотеки оно работало, а во второй нет. 

```
void handle_file(fb::Update& u) {
    String file_name = u.message().document().name();
    String chat_id = u.message().chat().id();
    Serial.printf("Downloading file: %s\n", file_name.c_str());
    
    if (file_name.endsWith(".bin")) {        
        bot.sendMessage(fb::Message("OTA update started", chat_id));
        fb::Fetcher fetch = bot.downloadFile(u.message().document().id());
        bool ok = fetch.updateFlash();
        bot.sendMessage(fb::Message(ok ? "OTA done" : "OTA error", chat_id));
    }
}
```
использую прям из примера код практически, все равно не хочет. OTA error и все.

woodlist, 2025-01-11:
Хороший вопрос, поддерживаю актуальность. Сам не проверял, собирался попросить автора библиотеки наделить такой возможностью код.

GyverLibs, 2025-01-11:
У меня работает отлично, у людей в чате тоже. Возможно не та разметка памяти

vvzvlad, 2025-01-11:
> У меня работает отлично, у людей в чате тоже. Возможно не та разметка памяти

А можете показать разметку, с которой работает? Я пробовал стандартную для OTA, с которой работает HTTP загрузка и прошлая версия библиотеки.

GyverLibs, 2025-01-11:
Я не менял разметку, стандартная у pio

woodlist, 2025-01-11:
Проверил сам, проблема есть, и она такая:
`Compilation error: 'class fb::Update' has no member named 'OTA'`

GyverLibs, 2025-01-11:
Ну а с чего проходить компиляции, если в библиотеке нет `isFile`

woodlist, 2025-01-11:
Да, действительно, убрал isFile и компиляция прошла успешно. Теперь буду испытать в железе.

woodlist, 2025-01-11:
Немного оффтопик, но все же... Попытался совершить ОТА на AI Thinker-CAM, что не выполнимо физический, так как при заливке кода этот борт требует закорочения на GPIO 0 на землю. Но чудо случилось в другом. До всех переделок скетча компилированный код занимал 83% флеш-памяти. После добавления строк для ОТА код стал занимать 35% памяти чипа.

GyverLibs, 2025-01-11:
А Иде не могла автоматически подставить другую разметку? Может размер бинаря в байтах посмотрим?

woodlist, 2025-01-11:
> А Иде не могла автоматически подставить другую разметку? Может размер бинаря в байтах посмотрим?

Бинарный весит 1.02Мб.

GyverLibs, 2025-01-11:
В обоих случаях?

woodlist, 2025-01-11:
Ну да. Флеш данного борда составляет 4Мб.

GyverLibs, 2025-01-11:
То есть прошивка занимает одинаковый размер в байтах но разный в процентах? Может проще уже приложить сюда оба лога компиляции

GyverLibs, 2025-01-11:
Поэтому я и попросил размеры бинаря в байтах в обоих случаях, потому что мы друг друга не понимаем