GyverLibs/FastBot2 #31: Не работает skipUpdates.

vvzvlad, 2024-12-29:
В setup делаю 
```
    bot.skipUpdates(-10);
    bot.attachUpdate(updateh);
    bot.setToken(bot_token);
    bot.setPollMode(fb::Poll::Long, 20000);
```
Выключаю устройство, пишу команду, включаю устройство, присылает ответ на команду. Правда, у меня он мониторит не личку, а группу, но насколько я понимаю, эта функция должна пропускать любые сообщения.

GyverLibs, 2024-12-29:
[Вот из доки](https://core.telegram.org/bots/api#getupdates)

> The negative offset can be specified to retrieve updates starting from -offset update from the end of the updates queue. All previous updates will be forgotten.

vvzvlad, 2024-12-29:
Все еще не понимаю. Я пробовал: -1, -10, 0, 10 — и все еще как минимум одно сообщение обрабатывается. Мне надо сделать простую вещь — сбросить очередь сообщений при запуске.

GyverLibs, 2024-12-30:
Можно сбросить все кроме последнего

vvzvlad, 2024-12-30:
> Можно сбросить все кроме последнего

в смысле, любая из "-1, -10, 0" сбрасывает? Тогда почему у меня продолжают приходить сообщения?

GyverLibs, 2024-12-30:
-1 сбрасывает все кроме последнего. Настойка применяется после следующего запроса getUpdates

vvzvlad, 2024-12-30:
Тогда как мне пропустить все, что есть на сервере? 

```
    bot.setPollMode(fb::Poll::Long, 20000);
    bot.skipUpdates(0);
    bot.getUpdates(true);
    bot.setToken(bot_token);
    bot.attachUpdate(updateh);
```
Так получается?

GyverLibs, 2024-12-30:
Нет, 0 не работает, нужен -1

vvzvlad, 2024-12-30:
```
    bot.setPollMode(fb::Poll::Long, 20000);
    bot.skipUpdates(-1);
    bot.getUpdates(true);
    bot.setToken(bot_token);
    bot.attachUpdate(updateh);
```
Оставляет одно сообщение, последнее. А как пропустить все?

GyverLibs, 2024-12-30:
Получается никак, надо игнорировать один апдейт

closed 2024-12-30 by vvzvlad