GyverLibs/FastBot2 #35: TextEdit не возвращает res.getRaw() в случае ошибки

JstSnt, 2025-03-27:
```
#define FB_USE_LOG Serial
#include <FastBot2.h>
bot.setToken(F(BOT_TOKEN));
bot.setPollMode(fb::Poll::Long, 20000);

void test_edit(){
    fb::TextEdit msg;
    msg.text = startupMessage;
    msg.chatID = CHAT_ID;
    msg.messageID = msgid;
    fb::Result res = bot.editText(msg);
    Serial.println(res.getRaw());
    res.reset();
}
```
если startupMessage не меняется, или сообщение с msgid удалили, то в serial пусто

``` 
> send + wait in sendPacket() [src/core/core.h : 350]
> got json in _parseResponse() [src/core/core.h : 439]

> getUpdates in getUpdates() [src/core/core.h : 290]
```

JstSnt, 2025-03-27:
может кому пригодится

в result.h
```cpp
bool parseJson() {
        if (!_reader) return 0;
        if (!_parser.parse(_reader.stream, _reader.length())) return 0;

        _parser.hashKeys();

        gson::Entry result = _parser[tg_apih::ok].toBool() 
          ? _parser[tg_apih::result] 
          : _parser.getByIndex(0);

        if (!result) return 0;
        *((gson::Entry*)this) = result;
        return 1;
    }
```

и в api.h добавить в namespace tg_apih

FB_MAKE_HASH(error_code)

и можно делать проверки

```
fb::Result res = bot.editText(msg); 
res[tg_apih::ok]
res[tg_apih::error_code]
res[tg_apih::description]
```

GyverLibs, 2025-06-02:
обработка ошибок всех запросов добавлена нативно, наверное больше не актуально

closed 2025-06-02 by GyverLibs