GyverLibs/FastBot2 #38: Странное поведение при попытке сохранить message().chat().id()

Olejan, 2025-04-24:
Мне нужно сохранить ChatId, чтобы использовать его в дальнейшем. Я написал код на основе минимального кода из примеров с сохранением chat_id в Preferences и восстановлением его при рестарте системы. Preferences отрабатывает как надо, но вот chat_id типа Value ведёт себя странно. В чём проявляется странность:
При первом запуске системы, когда ещё нет сохранённых данных в Preferences в глобальную переменную Value chat_id кладётся ноль, и это видно по выводу в порт. Когда я из Телеграма присылаю сообщение в чат "1", то видно, что chat_id равен нулю, поэтому приравнивается к u.message().chat().id(), и это видно через вывод в порт. Если я ещё раз пришлю любое сообщение в чат Телеграма, то будет видно, что chat_id уже имеет значение, поэтому пересохранения его не происходит.

В чём же странность? А вот она где. Если перезагрузить систему, то в порту будет видно, что chat_id восстановился из Preferences. Но если я пришлю любое сообщение в чат, то окажется, что chat_id почему-то равен нулю, поэтому он перезапишется из u.message().chat().id(). Далее всё будет работать нормально.

Если опять перезапустить систему, ситуация повторится. Кстати, попытка отправить сообщение админу с пустым id чата не приведёт к появлению сообщения в чате.

Использую esp32 v2.0.11. ArduinoIDE 2.3.5. StringUtils 1.4.30, FastBot2 1.1.0. Версия SDK не новая, но если я обновлюсь, перестанут компилироваться другие мои рабочие проекты. Не думаю, что версия SDK влияет на поведение класса Value, хотя я могу ошибаться.

```
#include <Arduino.h>

// Для проверки кода ssid, pass и token заполнить своими данными
#define WIFI_SSID "" 
#define WIFI_PASS ""
#define BOT_TOKEN ""
#define CHAT_ID ""

#include <FastBot2.h>
#include <Preferences.h>

Preferences prefs;
FastBot2 bot;
Value chat_id((int64_t)0);

void updateh(fb::Update& u) {
    Serial.println("NEW MESSAGE");
    Serial.println(u.message().from().username());
    Serial.println(u.message().text());

    Serial.print("chat_id before: ");
    Serial.println(chat_id);
    if (chat_id.toInt64() == 0)
    {
        chat_id = u.message().chat().id();
        int len = prefs.putLong64("chat_id", chat_id);
        Serial.print("Save chat_id: ");
        Serial.println(chat_id);
        Serial.print("Saved length: ");
        Serial.println(len);
    }

    bot.sendMessage(fb::Message(u.message().text(), u.message().chat().id()));
}

void setup() {
    Serial.begin(115200);
    Serial.println();
    WiFi.begin(WIFI_SSID, WIFI_PASS);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("Connected");
    bot.attachUpdate(updateh);   // подключить обработчик обновлений
    bot.setToken(F(BOT_TOKEN));  // установить токен
    bot.setPollMode(fb::Poll::Long, 20000);

    prefs.begin("Settings", false);
    chat_id = prefs.getLong64("chat_id", 0);
    printf("int64_t chat_id: %lld\n", chat_id);
    Serial.println(chat_id);
    printf("chat_id == 0 -> %s\n", chat_id.toInt64() == 0 ? "true" : "false");

    // поприветствуем админа
    bot.sendMessage(fb::Message("Hello!", CHAT_ID));
}

void loop() {
    // вызывать тикер в loop
    bot.tick();
}
```

Вывод в порт:

```
11:16:17.818 -> ...Connected
11:16:18.792 -> int64_t chat_id: 0
11:16:18.792 -> 0
11:16:18.792 -> chat_id == 0 -> true
11:16:33.725 -> NEW MESSAGE
11:16:33.725 -> nickname
11:16:33.725 -> 1
11:16:33.725 -> chat_id before: 
11:16:33.725 -> Save chat_id: 123458877
11:16:33.725 -> Saved length: 8
11:16:53.886 -> NEW MESSAGE
11:16:53.886 -> olejanja
11:16:53.886 -> 2
11:16:53.886 -> chat_id before: 123458877
11:17:03.389 -> ets Jun  8 2016 00:22:57
11:17:03.389 -> 
11:17:03.389 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:17:03.389 -> configsip: 0, SPIWP:0xee
11:17:03.389 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:17:03.432 -> mode:DIO, clock div:1
11:17:03.432 -> load:0x3fff0030,len:1344
11:17:03.432 -> load:0x40078000,len:13964
11:17:03.432 -> load:0x40080400,len:3600
11:17:03.432 -> entry 0x400805f0
11:17:03.738 -> 
11:17:04.361 -> ..Connected
11:17:04.832 -> int64_t chat_id: 984978877
11:17:04.832 -> 123458877
11:17:04.832 -> chat_id == 0 -> false
11:17:20.852 -> NEW MESSAGE
11:17:20.852 -> olejanja
11:17:20.852 -> 1
11:17:20.852 -> chat_id before: �8�?"
11:17:20.852 -> Save chat_id: 123458877
11:17:20.852 -> Saved length: 8
```

GyverLibs, 2025-04-24:
Value и не будет корректно сохраняться. Для хранения ID нужно использовать `int64_t`, об этом написано в документации с примерами

closed 2025-04-24 by GyverLibs

Olejan, 2025-04-24:
> Value и не будет корректно сохраняться. Для хранения ID нужно использовать `int64_t`, об этом написано в документации с примерами

Так я сохраняю в Preference не Value, а его int64_t значение. Потом из int64_t восстанавливаю Value. Проблема в том, что когда Value уже имеет нужное значение, полученное в setup(), то в колбэке updateh() оно опять не пойми что. Но это происходит ровно один раз.

GyverLibs, 2025-04-24:
ну так и не нужно использовать Value

Olejan, 2025-04-24:
> ну так и не нужно использовать Value

Я бы и не использовал Value, но при попытке создать InlineMenu с chat_id типа int64_t у меня меню не создаётся. Когда я создаю строку (char[]) на основе сохранённого идентификатора, тогда меню создаётся нормально. Мне не хотелось возиться со строкой, поэтому и решил использовать Value, т.к. его ожидает на входе InlineMenu.

GyverLibs, 2025-04-24:
я починил, теперь должно работать напрямую с int64. Нужно обновить stringutils

GyverLibs, 2025-04-24:
О, а микродрзинг может ещё лучше сделать?) ничего не принимаю, пока так работает